PaulStoffregen/Tlc5940 #11: Cannot use this lib with TimerOne

jcnesci, 2017-05-29:
Hi there,
First off, I'm sorry if this isn't the best place for this, but here goes.
I am simply trying to use this Tlc5940 library with the TimerOne library in the same Arduino sketch (Teensy 3.2, Teensyduino installed on OSX).

My Arduino sketch is as simple as:
```
#include "TimerOne.h"
#include "Tlc5940.h"

void setup() {}
void loop() {}
```

The error seems to be about multiple `ftm1_isr` definitions.
```
/var/folders/31/mc3khzyn4p76dfvl53dqbc5m0000gn/T/arduino_build_830505/libraries/Tlc5940/Tlc5940.cpp.o: In function `ftm1_isr':
/Applications/Arduino.app/Contents/Java/hardware/teensy/avr/libraries/Tlc5940/Tlc5940.cpp:76: multiple definition of `ftm1_isr'
/var/folders/31/mc3khzyn4p76dfvl53dqbc5m0000gn/T/arduino_build_830505/libraries/TimerOne/TimerOne.cpp.o:/Applications/Arduino.app/Contents/Java/hardware/teensy/avr/libraries/TimerOne/TimerOne.cpp:35: first defined here
/Applications/Arduino.app/Contents/Java/hardware/tools/arm/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: Disabling relaxation: it will not work with multiple definitions
collect2: error: ld returned 1 exit status
Error compiling for board Teensy 3.2 / 3.1.
```

I realize Tlc5940 uses timer1, though I'm not savvy enough to understand how. 
Is there a way to use timer1 thru the Tlc library instead of including it as its own library?
Or any other solution, of course, is welcome.

Thanks!

PaulStoffregen, 2017-05-29:
Use Timer3 or IntervalTimer instead.

For a more detailed answer, please post this question on the forum.

http://forum.pjrc.com/forums/2-Project-Guidance

I write detailed tech answers on the forum, so the info becomes a searchable public knowledge base.

closed 2017-05-29 by PaulStoffregen

jcnesci, 2017-05-30:
Thanks a lot!