PaulStoffregen/Tlc5940 #15: Cannot compile for Arduino Nono Every

dima-fedotovs, 2020-03-04:
### Description

Cannot compile for Arduino Nono Every

### Steps To Reproduce Problem

1. Update Arduino AVR Boards definition (just in case) in Tools -> Board -> Board Manager
2. Select "Arduino Nano Every" board in Tools -> Board
3. Open example project (BasicUse for instance)
4. Press "Verify"

### Hardware & Software

Board: Arduino Nano Every
Shields / modules used: raw TLC5940 chip only
Arduino IDE version: 1.8.12
Teensyduino version (if using Teensy)
Version info & package name (from Tools > Boards > Board Manager): 1.8.2 Arduino AVR Boards
Operating system & version: Windows 10
Any other software or hardware?

### Arduino Sketch

```cpp
#include "Tlc5940.h"
void setup(){}
void loop(){}
```

### Errors or Incorrect Output

```
In file included from C:\Users\dima\Documents\Arduino\libraries\Tlc5940/tlc_config.h:65:0,

                 from C:\Users\dima\Documents\Arduino\libraries\Tlc5940/Tlc5940.h:26,

                 from C:\Users\dima\AppData\Local\Temp\arduino_modified_sketch_137764\BasicUse.pde:1:

C:\Users\dima\Documents\Arduino\libraries\Tlc5940/pinouts/chip_includes.h:123:2: error: #error "Unknown Chip!"

 #error "Unknown Chip!"

  ^~~~~

exit status 1
Error compiling for board Arduino Nano Every.
```

francysuri, 2020-12-05:
Same here on Arduino nano IoT. 
`
In file included from /Users/xx/Documents/Arduino/libraries/Tlc5940/tlc_config.h:65:0,
                 from /Users/xx/Documents/Arduino/libraries/Tlc5940/Tlc5940.h:26,
                 from /Users/xx/Documents/Arduino/love/love.ino:11:
/Users/xx/Documents/Arduino/libraries/Tlc5940/pinouts/chip_includes.h:123:2: error: #error "Unknown Chip!"
 #error "Unknown Chip!"
  ^~~~~
exit status 1
Errore durante la compilazione per la scheda Arduino NANO 33 IoT.`
This library needs an update. I don't know the structure of this but if I can be of help, I will gladly do it.

ScharreSoft, 2021-03-08:
Same here using two daisy chained TLC5940's on Nano Every. My sketch works fine on an Arduino Nano v3 but does not compile when board is set to Arduino Nano Every. 

Using Arduino IDE version: 1.8.13 I get the same error message as above.

<code>
In file included from C:\Users\Willem\Documents\Arduino\libraries\Tlc5940/tlc_config.h:65:0,
                 from C:\Users\Willem\Documents\Arduino\libraries\Tlc5940/Tlc5940.h:26,
                 from C:\Users\Willem\Desktop\Arduino\16_SmartStep_v3\16_SmartStep_v3.ino:3:
C:\Users\Willem\Documents\Arduino\libraries\Tlc5940/pinouts/chip_includes.h:123:2: error: #error "Unknown Chip!"
 #error "Unknown Chip!"
  ^~~~~
exit status 1
Fout bij het compileren voor board Arduino Nano Every
</code>

I am not skilled enough to update the library, however, I could be of help in testing new code.

ScharreSoft, 2021-05-14:
I have made a TLC5940 library that can be used with the Arduino Nano Every. It can be found here: https://github.com/ScharreSoft/Tlc5940_AVR0