PaulStoffregen/Tlc5940 #19: Arduino Nano 33 IoT

97Cweb, 2022-05-21:
### Description

Hello all, I am currently upgrading my project from an Arduino Nano to a Nano 33 IoT board, which is supposed to be just a drop in replacement. I am testing it currently, and this library throws an error regarding missing board type when compiling. 

Hopefully there is enough information to resolve this, if I am missing anything pleas let me know. I finally have motivation to work on this project again, and it hinges on this still working for this new board. From what I read, both nanos should be drop in replacements of each other. Don't know if that helps you though. 

### Steps To Reproduce Problem

Create a new sketch, Target IoT board in board selector. Include this library, Hit checkmark. 

### Hardware & Software

Board: Arduino Nano 33 IoT
Shields / modules used : none
Arduino IDE version: 1.8.19
Teensyduino version (if using Teensy) Not used
Version info & package name (from Tools > Boards > Board Manager) Arduino SAMD Boards (32-bits ARM Cortex-M0+) by Arduino version 1.8.13
Operating system & version Windows 10
Any other software or hardware? Dell laptop to run Arduino IDE in

### Arduino Sketch

```
// libraries: just Tlc5940.h Freshly downloaded and unzipped into library folder of Arduino as of today. Only change is to make config take 3 chips instead of 1. 

#include "Tlc5940.h"

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Errors or Incorrect Output

`Arduino: 1.8.19 (Windows 10), Board: "Arduino NANO 33 IoT"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\97cwe\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\97cwe\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\97cwe\Documents\Arduino\libraries -fqbn=arduino:samd:nano_33_iot -vid-pid=2341_8057 -ide-version=10819 -build-path C:\Users\97cwe\AppData\Local\Temp\arduino_build_582725 -warnings=none -build-cache C:\Users\97cwe\AppData\Local\Temp\arduino_cache_389484 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\97cwe\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Users\97cwe\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.CMSIS-Atmel.path=C:\Users\97cwe\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.CMSIS-Atmel-1.2.0.path=C:\Users\97cwe\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.CMSIS.path=C:\Users\97cwe\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.CMSIS-4.5.0.path=C:\Users\97cwe\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.bossac.path=C:\Users\97cwe\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0-arduino3 -prefs=runtime.tools.bossac-1.7.0-arduino3.path=C:\Users\97cwe\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0-arduino3 -prefs=runtime.tools.openocd.path=C:\Users\97cwe\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino7 -prefs=runtime.tools.openocd-0.10.0-arduino7.path=C:\Users\97cwe\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino7 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\97cwe\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\97cwe\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -verbose C:\Users\97cwe\Documents\Computer Stuff\Arduino\Projects\LED Cube\fix_TLC5940_library\fix_TLC5940_library.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\97cwe\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\97cwe\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\97cwe\Documents\Arduino\libraries -fqbn=arduino:samd:nano_33_iot -vid-pid=2341_8057 -ide-version=10819 -build-path C:\Users\97cwe\AppData\Local\Temp\arduino_build_582725 -warnings=none -build-cache C:\Users\97cwe\AppData\Local\Temp\arduino_cache_389484 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\97cwe\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Users\97cwe\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.CMSIS-Atmel.path=C:\Users\97cwe\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.CMSIS-Atmel-1.2.0.path=C:\Users\97cwe\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.CMSIS.path=C:\Users\97cwe\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.CMSIS-4.5.0.path=C:\Users\97cwe\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.bossac.path=C:\Users\97cwe\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0-arduino3 -prefs=runtime.tools.bossac-1.7.0-arduino3.path=C:\Users\97cwe\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0-arduino3 -prefs=runtime.tools.openocd.path=C:\Users\97cwe\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino7 -prefs=runtime.tools.openocd-0.10.0-arduino7.path=C:\Users\97cwe\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino7 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\97cwe\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\97cwe\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -verbose C:\Users\97cwe\Documents\Computer Stuff\Arduino\Projects\LED Cube\fix_TLC5940_library\fix_TLC5940_library.ino

Using board 'nano_33_iot' from platform in folder: C:\Users\97cwe\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.13

Using core 'arduino' from platform in folder: C:\Users\97cwe\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.13

Detecting libraries used...

"C:\\Users\\97cwe\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10819 -DARDUINO_SAMD_NANO_33_IOT -DARDUINO_ARCH_SAMD -DCRYSTALLESS -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x8057 -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino NANO 33 IoT\"" "-IC:\\Users\\97cwe\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS\\4.5.0/CMSIS/Include/" "-IC:\\Users\\97cwe\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS-Atmel\\1.2.0/CMSIS/Device/ATMEL/" "-IC:\\Users\\97cwe\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.13\\cores\\arduino/api/deprecated" "-IC:\\Users\\97cwe\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.13\\cores\\arduino/api/deprecated-avr-comp" "-IC:\\Users\\97cwe\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.13\\cores\\arduino" "-IC:\\Users\\97cwe\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.13\\variants\\nano_33_iot" "C:\\Users\\97cwe\\AppData\\Local\\Temp\\arduino_build_582725\\sketch\\fix_TLC5940_library.ino.cpp" -o nul

Alternatives for Tlc5940.h: [Tlc5940@0.15]

ResolveLibrary(Tlc5940.h)

  -> candidates: [Tlc5940@0.15]

"C:\\Users\\97cwe\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10819 -DARDUINO_SAMD_NANO_33_IOT -DARDUINO_ARCH_SAMD -DCRYSTALLESS -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x8057 -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino NANO 33 IoT\"" "-IC:\\Users [... cut ...]
                 from C:\Users\97cwe\Documents\Arduino\libraries\Tlc5940/Tlc5940.h:26,

                 from C:\Users\97cwe\Documents\Computer Stuff\Arduino\Projects\LED Cube\fix_TLC5940_library\fix_TLC5940_library.ino:1:

C:\Users\97cwe\Documents\Arduino\libraries\Tlc5940/pinouts/chip_includes.h:129:2: error: #error "Unknown Chip!"

 #error "Unknown Chip!"

  ^~~~~

Using library Tlc5940 at version 0.15 in folder: C:\Users\97cwe\Documents\Arduino\libraries\Tlc5940 

exit status 1

Error compiling for board Arduino NANO 33 IoT.

`

PaulStoffregen, 2022-05-21:
Nano 33 is a completely different processor with different timer hardware than regular Nano.

97Cweb, 2022-05-21:
Hello Paul.
Thank you for your reply. Does this mean, it will take a while, or that you will not be upgrading this system for this newer Arduino board?

PaulStoffregen, 2022-05-21:
It will only happen if someone who uses Nano 33 and knows its timers well enough contributes the code.  I will not do it.  Depending on the open source community for those sorts of contributions.

97Cweb, 2022-05-21:
Ok, thank you for being open and honest about this. Hopefully someone eventually adds it.

97Cweb, 2022-06-16:
Hi Paul
I did manage to sort it out partly. It is no good enough to go into the library, but I did manage to get it to work for my system. I am giving it to you so whoever wants to take a crack at it can, as you are the main contact for this library. https://drive.google.com/file/d/1FIm5III9k6nUCRf4pHXzgGrBw8DfuRJw/view?usp=sharing

PaulStoffregen, 2022-06-16:
Any chance you could upload the changed code here on Github?

Long-term, odds are someone else will find this message, but Google Drive tends to not keep files for long term storage.

PaulStoffregen, 2022-06-16:
Looked at this briefly.  Seems you probably stated with an older copy of this library, from around 2016.

Here's my best attempt to capture the changes in diff format.... for future reference.

```diff
diff -r -u -N -w -B Tlc5940_old/pinouts/Arduino_Nano_33_IOT.h Tlc5940/pinouts/Arduino_Nano_33_IOT.h
--- Tlc5940_old/pinouts/Arduino_Nano_33_IOT.h	1969-12-31 16:00:00.000000000 -0800
+++ Tlc5940/pinouts/Arduino_Nano_33_IOT.h	2022-06-11 22:01:36.000000000 -0700
@@ -0,0 +1,86 @@
+/*  Copyright (c) 2009 by Alex Leone <acleone ~AT~ gmail.com>
+
+    This file is part of the Arduino TLC5940 Library.
+
+    The Arduino TLC5940 Library is free software: you can redistribute it
+    and/or modify it under the terms of the GNU General Public License as
+    published by the Free Software Foundation, either version 3 of the
+    License, or (at your option) any later version.
+
+    The Arduino TLC5940 Library is distributed in the hope that it will be
+    useful, but WITHOUT ANY WARRANTY; without even the implied warranty of
+    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
+    GNU General Public License for more details.
+
+    You should have received a copy of the GNU General Public License
+    along with The Arduino TLC5940 Library.  If not, see
+    <http://www.gnu.org/licenses/>. */
+
+#ifndef TLC_Arduino_Nano_33_IOT_H
+#define TLC_Arduino_Nano_33_IOT_H
+
+// bitbang I/O is plenty fast on Nano 33
+#ifdef DATA_TRANSFER_MODE
+#undef DATA_TRANSFER_MODE
+#endif
+#define DATA_TRANSFER_MODE TLC_BITBANG
+
+/** \file
+    SPI and timer pins for the ATmega168/48/88.  Don't edit these.  All
+    changeable pins are defined in tlc_config.h */
+
+/** VPRG (Arduino digital pin 8) -> VPRG (TLC pin 27) */
+#define DEFAULT_VPRG_PIN    8
+#define DEFAULT_VPRG_PORT   REG_PORT_OUT0 
+#define DEFAULT_VPRG_DDR    REG_PORT_DIR0 
+
+/** XERR (Arduino digital pin 12) -> XERR (TLC pin 16) */
+#define DEFAULT_XERR_PIN    12
+#define DEFAULT_XERR_PORT   REG_PORT_OUT0 
+#define DEFAULT_XERR_DDR    REG_PORT_DIR0 
+#define DEFAULT_XERR_PINS   12
+
+/** SIN (Arduino digital pin 11) -> SIN (TLC pin 26) */
+#define DEFAULT_BB_SIN_PIN      11
+#define DEFAULT_BB_SIN_PORT     REG_PORT_OUT0 
+#define DEFAULT_BB_SIN_DDR      REG_PORT_DIR0 
+/** SCLK (Arduino digital pin 13) -> SCLK (TLC pin 25) */
+#define DEFAULT_BB_SCLK_PIN     13
+#define DEFAULT_BB_SCLK_PORT    REG_PORT_OUT0 
+#define DEFAULT_BB_SCLK_DDR     REG_PORT_DIR0 
+
+/** MOSI (Arduino digital pin 11) -> SIN (TLC pin 26) */
+#define TLC_MOSI_PIN     11
+#define TLC_MOSI_PORT    REG_PORT_OUT0 
+#define TLC_MOSI_DDR     REG_PORT_DIR0 
+
+/** SCK (Arduino digital pin 13) -> SCLK (TLC pin 25) */
+#define TLC_SCK_PIN      13
+#define TLC_SCK_PORT     REG_PORT_OUT0 
+#define TLC_SCK_DDR      REG_PORT_DIR0 
+
[... 262 lines cut ...]
 /** Determines how data should be transfered to the TLCs.  Bit-banging can use
     any two i/o pins, but the hardware SPI is faster.
@@ -91,15 +91,17 @@
     \f$\displaystyle TLC\_PWM\_PERIOD =
        \frac{(TLC\_GSCLK\_PERIOD + 1) * 4096}{2} \f$
     \note The default of 8192 means the PWM frequency is 976.5625Hz */
-#define TLC_PWM_PERIOD    8192
-
+//#define TLC_PWM_PERIOD    8192
+#define TLC_PWM_PERIOD   16384 
+//#define TLC_PWM_PERIOD   14336 
 /** Determines how long each period GSCLK is.
     This is related to TLC_PWM_PERIOD:
     \f$\displaystyle TLC\_GSCLK\_PERIOD =
        \frac{2 * TLC\_PWM\_PERIOD}{4096} - 1 \f$
     \note Default is 3 */
-#define TLC_GSCLK_PERIOD    3
-
+//#define TLC_GSCLK_PERIOD    3
+#define TLC_GSCLK_PERIOD    7
+//#define TLC_GSCLK_PERIOD    6
 /** Enables/disables VPRG (TLC pin 27) functionality.  If you need to set dot
     correction data, this needs to be enabled.
     - 0 VPRG is not connected.  <em>TLC pin 27 must be grounded!</em> (default)
```

97Cweb, 2022-06-16:
https://github.com/97Cweb/TLC5940-Nano-33
Here is the Github copy of it for others

97Cweb, 2022-10-11:
I can upload it. Do I make it in my own, or is there a way to tie it to
yours?