PaulStoffregen/Tlc5940 #23: Compatibility with ATtiny 202/402?

CSHaze, 2023-05-27:
### Description

Is there a way to modify this to work with an ATtiny202/402? I am struggling to find equivalent timer settings. Are there any plans to expand this to work with other boards?

### Hardware & Software

Board = ATtiny 402
Software = Microchip studio XC8 compiler
Programmer = Amtel ICE

### Errors or Incorrect Output

I cannot seem to find the timer settings to get the SPI synced with XLAT.

PaulStoffregen, 2023-05-27:
No, I do not have plans to port to any other chips.  So far (~10 years), nobody else has shown any interest in porting or maintaining this old Arduino library.

CSHaze, 2023-05-27:
No worries, thanks for the quick response. If I get it working maybe I’ll do a pull request.

closed 2023-05-27 by CSHaze